slipx06/Sunsynk-Home-Assistant-Dash #6: Inverter Settings in Voltage

Gavrik13, 2023-11-20:
Hi!
How to read and write inverter settings when the settings are in volts, not percentages?
The battery is not connected to the inverter via CAN, but is controlled by voltage.
[image]

slipx06, 2023-11-20:
Hi. I will add it to the config.

slipx06, 2023-11-21:
@Gavrik13 I've added those sensors. Can you please test

`number.sunsynk_prog1_voltage`
`number.sunsynk_prog2_voltage`
`number.sunsynk_prog3_voltage`
`number.sunsynk_prog4_voltage`
`number.sunsynk_prog5_voltage`
`number.sunsynk_prog6_voltage`

Gavrik13, 2023-11-21:
Great ! 

[image]

closed 2023-11-21 by slipx06