slipx06/Sunsynk-Home-Assistant-Dash #8: Resizable Card

iendicott, 2024-02-16:
Hi,

Not sure if this has been requested before, but is there any chance you can make the SunSynk solar card resizeable please?

Regards Ivan

slipx06, 2024-02-16:
Hi. The card is resizable with some limitation. You have to switch to the code editor and add the following attributes

```
card_height: | Optional | 396px | Only used when panel_mode: false. Sets the card height in pixels. Specify the value i.e. 400px or provide a sensor i.e. input.numer_height
card_width: | Optional | 100% | Only used when panel_mode: true. Sets the card width in pixels or percentage 400px or 80%. For adjustments when using the Panel(1 card) view types or grid layouts
panel_mode: | Optional | false | Toggles panel mode setting card height to 100%. For use with Panel(1 card) view types or grid layouts
```

iendicott, 2024-02-16:
Thanks Slipx06 perfect. Happy with the Yaml code editor. 👍

closed 2024-02-16 by iendicott