slipx06/Sunsynk-Home-Assistant-Dash #11: Program Size too Large

hectorvanzyl, 2024-04-18:
Hi, when trying to write code to ESP32 I get following error after multple attempts and removing unneccessary lines not required for Sunsynk.
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
RAM:   [==        ]  18.1% (used 59444 bytes from 327680 bytes)
Flash: [==========]  104.6% (used 1919297 bytes from 1835008 bytes)
Error: The program size (1919297 bytes) is greater than maximum allowed (1835008 bytes)
*** [checkprogsize] Explicit exit, status 1

slipx06, 2024-04-18:
Hi. Have you tried to disable the webserver?

hectorvanzyl, 2024-04-18:
Hi, yes webserver entry was completely removed from yaml.

slipx06, 2024-04-18:
Not sure what the issue is. Did you try a clean build? This is mine with everything enabled

```
INFO ESPHome 2024.3.1
INFO Reading configuration /config/esphome/inverter.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing inverter (board: nodemcu-32s; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.3
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
RAM:   [=         ]  14.4% (used 47256 bytes from 327680 bytes)
Flash: [=======   ]  68.1% (used 1249745 bytes from 1835008 bytes)
========================= [SUCCESS] Took 15.29 seconds =========================
INFO Successfully compiled program.
```

hectorvanzyl, 2024-04-18:
I did try clean build, also did a "python -m esptool --chip esp32 erase_flash" to ensure memory have no leftovers, installed ESPHome and tried to upload code, starting to have my doubts regarding the dev board as the size discrepency do not make sense.

hectorvanzyl, 2024-04-18:
just started from scratch again, still geting same error, is there perhaps anything in the dependency graph that could cause the issue, if so how  do I go about to remove it.
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.3
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|-- Improv @ 1.2.3
RAM:   [==        ]  18.1% (used 59444 bytes from 327680 bytes)
Flash: [==========]  104.6% (used 1919297 bytes from 1835008 bytes)
Error: The program size (1919297 bytes) is greater than maximum allowed (1835008 bytes)

slipx06, 2024-04-18:
Are you using the correct board 

```
esp32:
  board: esp32dev
  framework:
    type: arduino
```

hectorvanzyl, 2024-04-18:
Exactly as above yes.

hectorvanzyl, 2024-04-18:
I think I'm wasting your time with an issue with the board, it's one of the 1st ESP32 dev boards released, I'm waiting for new ones.
Thanks for your time and brilliant code shared, it's truly appreciated.
Best wishes.

closed 2024-06-16 by slipx06