slipx06/Sunsynk-Home-Assistant-Dash #12: What are the names of the sensors that i need to use?

gohigher14, 2024-04-22:
i'm using this configuration : https://github.com/slipx06/Sunsynk-Home-Assistant-Dash/blob/main/ESPHome-1P-Sunsynk-Deye-Master-Slave.yaml

What i want to know is , what are the names of the 2 sensors i need to count together , to know my live home usage?

For example , my car is charging and uses aroud 2300w. The rest of the use in the house is now normally arround 700w. So this is a total of 3000 w.
if i count these 2 sensors together : sensor.deye_inverter_power and  sensor.deye_slave_inverter_power i get arround 2866w ( sometimes suddenly arround 5000w and sometimes it drops below 2000w)
if i count these 2 sensors together : sensor.deye_load_power and  sensor.deye_slave_load_power i get arround 3272w ( sometimes suddenly lower then 1200w)
So what are my correct sensors to choose from , so i know what my live home usage is?

slipx06, 2024-04-22:
HI
Assume that everything is connected on the essential side you can look at register 178 (Load Power) on both the primary and secondary inverters. 

You can also look at register 176 (Load L1 Power) but this should be the same

gohigher14, 2024-04-22:
> register 178

correct. Those are the ones i'm using now. But it's not giving me the correct values like you can see.
[image]
[image]

slipx06, 2024-04-22:
I don't know why those sensors show as 0? Did you check developer tools and state history?

gohigher14, 2024-04-22:
> I don't know why those sensors show as 0? Did you check developer tools and state history?

i'm getting some weird results from this sensor 
[image]

This sensor is better

[image]

slipx06, 2024-04-22:
Could be related to your integration. There was another user that experienced something similar here https://powerforum.co.za/topic/22487-8kw-sunsynk-with-esp32-and-rs485-not-communicating/?do=findComment&comment=202292

Things to check

1. Power supply to rs485 and ESP
2. Wiring
3. Reboot Inverter

Have you checked any of the other sensors i.e. frequency, solar power etc for anomalies, strange values?

gohigher14, 2024-04-22:
> Could be related to your integration. There was another user that experienced something similar here [...]

[image]

Like you can see , my frequency is ok. Should be around 50 Hz and that's ok.

It can be that is has someting to do with my setup maybe? I have 2 single fase inverters connected on my 3 fase grid. But here we have a delta grid. That's 3 fases without a common N. Between each fase 230v. Inverter 1 is connected on fase 1 and fase 2. And other inverter is connected between fase 2 and fase 3. Have no other solutions here's because it's not allowed to have 3 inverters on this 3 fase delta grid. Also can only put a max of 5kW per fase , with a total of 10kW or only 2 inverters.