slipx06/Sunsynk-Home-Assistant-Dash #13: Serial number not displaying correctly

ameran, 2024-05-01:
Hi, 

When copying the code to create an entity for serial number, when I use this code:
  - platform: modbus_controller            # 003-007 Inverter HEX Serial Number
    modbus_controller_id: inverter
    id: inverter_esphome_serial
    register_type: holding
    skip_updates: ${settings_skipped_updates}
    raw_encode: HEXBYTES
    address: 003
    register_count: 5
    internal: true

  - platform: template                     # Inverter ASCII Serial Number
    name: "${friendly_name} Serial"
    id: inverter_esphome_ascii_serial
    icon: "mdi:text"
    lambda: |-
      std::string hexData = id(inverter_esphome_serial).state;

      // Convert hex-encoded string to ASCII
      std::string asciiData;
      for (size_t i = 0; i < hexData.length(); i += 2) {
        std::string byte = hexData.substr(i, 2);
        char chr = static_cast<char>(std::stoi(byte, nullptr, 16));
        asciiData.push_back(chr);
      }

      return asciiData;

mine only displays a single registers value.
Its like its not combining the registers, is there something I'm doing wrong?

slipx06, 2024-05-01:
@ameran thanks for reporting. I was missing the response_size when reading the registers. See updated code below

```
  - platform: modbus_controller            # 003-007 Inverter HEX Serial Number
    modbus_controller_id: inverter
    id: inverter_esphome_serial
    register_type: holding
    skip_updates: ${settings_skipped_updates}
    raw_encode: HEXBYTES
    address: 003
    register_count: 5
    response_size: 9
    internal: true
```

ameran, 2024-05-02:
Awesome!
It's working perfectly!

closed 2024-05-02 by slipx06

ameran, 2024-05-03:
Quick note, this line should be:
response_size: 10
that one then matches whats on the display.

slipx06, 2024-05-03:
Thanks also realized this.