slipx06/Sunsynk-Home-Assistant-Dash #14: Control board & Comms board firmware

ameran, 2024-05-02:
Not so much an issue, but can you look into the code for reading the firmware versions for the Controller board and the communications board's firmware?
Address 013 & 014

I've tried the below, but it doesn't work:
  - platform: modbus_controller            # 013 Control board firmware
    modbus_controller_id: deye
    name: "Controller Board Firmware"
    id: deye_controlboard_firmware
    icon: "mdi:text"
    register_type: holding
    raw_encode: HEXBYTES
    address: 013
    register_count: 5
    response_size: 9
    internal: true
    
    
  - platform: modbus_controller            # 014 Comms board firmware
    modbus_controller_id: deye
    name: "Communications Board Firmware"
    id: deye_communicationboard_firmware
    icon: "mdi:text"
    register_type: holding
    raw_encode: HEXBYTES
    address: 014
    register_count: 5
    response_size: 9
    internal: true

slipx06, 2024-05-02:
@ameran  Try this and share the output

```
platform: modbus_controller # 013 Control board firmware
modbus_controller_id: deye
name: "Controller Board Firmware"
id: deye_controlboard_firmware
icon: "mdi:text"
register_type: holding
raw_encode: HEXBYTES
address: 013
internal: false

platform: modbus_controller # 014 Comms board firmware
modbus_controller_id: deye
name: "Communications Board Firmware"
id: deye_communicationboard_firmware
icon: "mdi:text"
register_type: holding
raw_encode: HEXBYTES
address: 014
internal: false
```

ameran, 2024-05-03:
Work, Thanks!

slipx06, 2024-05-03:
I'm not sure. The docs only give one 16 bit register to read so I don't know where the rest of the data is stored

slipx06, 2024-05-03:
You can try read the adjacent registers and see what you get

```
platform: modbus_controller # 013 Control board firmware
modbus_controller_id: deye
name: "Controller Board Firmware"
id: deye_controlboard_firmware
icon: "mdi:text"
register_type: holding
raw_encode: HEXBYTES
address: 012
internal: false
register_count: 2
response_size: 2

platform: modbus_controller # 014 Comms board firmware
modbus_controller_id: deye
name: "Communications Board Firmware"
id: deye_communicationboard_firmware
icon: "mdi:text"
register_type: holding
raw_encode: HEXBYTES
address: 014
internal: false
register_count: 2
response_size: 2
```

ameran, 2024-05-03:
Thanks, 

I tried that, modified it a bit as it wasn't pulling much more.
Used this:
 - platform: modbus_controller # 013 Control board firmware 
    modbus_controller_id: deye
    name: "Controller Board Firmware"
    id: deye_controlboard_firmware
    icon: "mdi:text"
    register_type: holding
    raw_encode: HEXBYTES
    address: 013
    internal: false
    response_size: 8

  - platform: modbus_controller # 014 Comms board firmware
    modbus_controller_id: deye
    name: "Communications Board Firmware"
    id: deye_communicationboard_firmware
    icon: "mdi:text"
    register_type: holding
    raw_encode: HEXBYTES
    address: 014
    internal: false
    response_size: 8

now I get this:
[image]
[image]

which isn't too wrong, its just looks like its in the wrong order
/Users/ameetranchod/Downloads/IMG_8627.jpg

slipx06, 2024-05-03:
Maybe I'm understanding this wrong but according to your output.

```
register 13 = 1515
register 14 = 0000
Register 15 = 3373
Register 16 = c356
```

```
1   2  3  4   5  6   7  8
1515 0000 3373 c356
0000 3373 c356 0000
```

slipx06, 2024-05-03:
Try this

```
- platform: modbus_controller
  modbus_controller_id: deye
  id: deye_communicationboard_firmware_14
  icon: "mdi:text"
  register_type: holding
  raw_encode: HEXBYTES
  address: 014
  internal: true

- platform: modbus_controller
  modbus_controller_id: deye
  id: deye_communicationboard_firmware_16
  icon: "mdi:text"
  register_type: holding
  raw_encode: HEXBYTES
  address: 016
  internal: true

- platform: template
  name: "Communications Board Firmware"
  id: inverter_esphome_firmware
  icon: "mdi:text"
  lambda: |-
    return id(deye_communicationboard_firmware_14).state + "-" + id(deye_communicationboard_firmware_16).state;

- platform: modbus_controller
  modbus_controller_id: deye
  id: deye_controlboard_firmware_15
  icon: "mdi:text"
  register_type: holding
  raw_encode: HEXBYTES
  address: 015
  internal: true

- platform: modbus_controller
  modbus_controller_id: deye
  id: deye_controlboard_firmware_13
  icon: "mdi:text"
  register_type: holding
  raw_encode: HEXBYTES
  address: 013
  internal: true

- platform: template
  name: "Controller Board Firmware"
  id: inverter_esphome_firmware
  icon: "mdi:text"
  lambda: |-
    return id(deye_controlboard_firmware_15).state + "-" + id(deye_controlboard_firmware_13).state;

```

slipx06, 2024-05-03:
For the sunsynk

```
  - platform: modbus_controller            # 013 Software Firmware Version
    modbus_controller_id: inverter
    id: inverter_esphome_soft_version_13
    name: "${friendly_name} Software Version"
    register_type: holding
    skip_updates: ${settings_skipped_updates}
    raw_encode: HEXBYTES
    icon: "mdi:text"
    address: 013

  - platform: modbus_controller            # 015 Inverter Firmware Version
    modbus_controller_id: inverter
    id: inverter_esphome_mcu_inverter_firmware_15
    name: "${friendly_name} MCU Version"
    register_type: holding
    skip_updates: ${settings_skipped_updates}
    raw_encode: HEXBYTES
    icon: "mdi:text"
    address: 015

  - platform: modbus_controller            # 016 Inverter LCD Firmware Version
    modbus_controller_id: inverter
    id: inverter_esphome_comm_lcd_firmware_16
    name: "${friendly_name} COMM Version"
    register_type: holding
    skip_updates: ${settings_skipped_updates}
    raw_encode: HEXBYTES
    icon: "mdi:text"
    address: 016
```

slipx06, 2024-05-03:
Finally settled on

```
  - platform: modbus_controller            # 013 Inverter MCU Firmware Version
    modbus_controller_id: inverter
    id: inverter_esphome_MCU_version_13
    register_type: holding
    skip_updates: ${settings_skipped_updates}
    raw_encode: HEXBYTES
    address: 013
    internal: true

  - platform: modbus_controller            # 015 Inverter MCU Firmware Version
    modbus_controller_id: inverter
    id: inverter_esphome_MCU_version_15
    register_type: holding
    skip_updates: ${settings_skipped_updates}
    raw_encode: HEXBYTES
    address: 015
    internal: true

  - platform: template                      # 015-013 Inverter MCU version
    name: "${friendly_name} MCU Version"
    id: inverter_esphome_main_version
    icon: "mdi:text"
    lambda: |-
      return id( inverter_esphome_MCU_version_15).state + "-" + id(inverter_esphome_MCU_version_13).state;

  - platform: modbus_controller            # 016 Inverter LCD Firmware Version
    modbus_controller_id: inverter
    id: inverter_esphome_comm_lcd_firmware_16
    name: "${friendly_name} COMM Version"
    register_type: holding
    skip_updates: ${settings_skipped_updates}
    raw_encode: HEXBYTES
    icon: "mdi:text"
    address: 016
```

ameran, 2024-05-04:
> Finally settled on
>
> ```
>   - platform: modbus_controller            # 013 Inverter MCU Firmware Version
>     modbus_controller_id: inverter
>     id: inverter_esphome_MCU_version_13
>     register_type: [...]

PERFECT!
See the results below:
[image]

closed 2024-05-04 by ameran