slipx06/Sunsynk-Home-Assistant-Dash #20: Unable to write or read settings from inverter after firmware update.

ameran, 2024-08-16:
I've had my system running for months now and it's been fine.

We had Deye upgrade the firmware of the unit because it wasn't recognising the new battery. After that the ESP would connect and read, but all the data had NaN values.

Dug around on the inverter itself and the Modbus was changed from 01 to 00, must have happened when the new firmware was installed. Changed that in the code and it started reading again.

Now the problem is this:

[image]

It's not reading the time zone values, only displaying 21474 and also not displaying the SOC% only NaN.

What do I do to fix this?

Example of the Write settings:

[image]

[image]
[image]
[image]

slipx06, 2024-08-16:
Hi. Have you looked at your ESPHome logs?

ameran, 2024-08-16:
Yip, no errors, only this:
[image]
[image]

It was all working 100% before we did the firmware upgrade, since then it seems like registers have changed.
I'd love to know where to start looking.

slipx06, 2024-08-16:
Looks like it's not reading any of the System Timer info?
You can try read the raw register values and see what comes back.

Registers 
250-255 - Times
274-279 - Charge
268-273 - SOC
256-251 - Power

Have you also checked to make sure there are no other errors for the other registers?

ameran, 2024-08-16:
i'm clueless.

The other registers are definitely pulling values:
[image]
It's just the ones to do with the timer info.

I've reverted to your code and only changed the details relevant to get it connected to my network and the inverter, incase I have a typo somewhere, but its the same.
The same registers come back with the strange values or unknown.

Do you think the inverter address could be an issue?
before it was 0x01, its now 0x00.

ameran, 2024-08-16:
I fixed it, it wasn't anything in the code!
It was the inverter, when it was reset the modbus SN setting was set back to 00, I changed it to 01 and its reading all the settings again.

closed 2024-08-16 by ameran