slipx06/Sunsynk-Home-Assistant-Dash #22: Using Solcast insead of Meteo Solar Forecast

Ben00793, 2024-09-03:
Hello,
I was just wondering if it would be possible to use Solcast sensors insead of Meteo Solar Forecast in this card:
[image]

if yes could you please help me with what do I need to change in the code?
Thank you!

Ben00793, 2024-09-04:
So far this is how far I got:
type: custom:plotly-graph
entities:
  - entity: sensor.solcast_pv_forecast_forecast_today
    name: PV Forecast
    line:
      color: rgb(125, 125, 125)
      shape: spline
    fill: tozeroy
    filters:
      - fn: |
          ({meta}) => {
            const forecast = meta.DetailedForecast;
            const xs = forecast.map(item => new Date(item.period_start));
            const ys = forecast.map(item => item.pv_estimate);
            return { xs, ys };
          }
    yaxis: y1
    showlegend: true

It gives me this error: Error: at [entities.0.filters.0]: Error in filter: TypeError: Cannot read properties of undefined (reading 'map')
Attributes of sensor.solcast_pv_forecast_forecast_today:
Attributes
Dayname
Wednesday
DataCorrect
true
DetailedForecast
- period_start: '2024-09-04T00:00:00+01:00'
pv_estimate: 0
pv_estimate10: 0
pv_estimate90: 0
- period_start: '2024-09-04T00:30:00+01:00'
pv_estimate: 0
pv_estimate10: 0
pv_estimate90: 0
- period_start: '2024-09-04T01:00:00+01:00'
pv_estimate: 0
pv_estimate10: 0
pv_estimate90: 0
- period_start: '2024-09-04T01:30:00+01:00'
pv_estimate: 0
pv_estimate10: 0
pv_estimate90: 0
- period_start: '2024-09-04T02:00:00+01:00'
pv_estimate: 0
pv_estimate10: 0
pv_estimate90: 0
- period_start: '2024-09-04T02:30:00+01:00'
pv_estimate: 0
pv_estimate10: 0
pv_estimate90: 0
- period_start: '2024-09-04T03:00:00+01:00'
pv_estimate: 0
pv_estimate10: 0
pv_estimate90: 0
- period_start: '2024-09-04T03:30:00+01:00'
pv_estimate: 0
pv_estimate10: 0
pv_estimate90: 0
- period_start: '2024-09-04T04:00:00+01:00'
pv_estimate: 0
pv_estimate10: 0
pv_estimate90: 0
- period_start: '2024-09-04T04:30:00+01:00'
pv_estimate: 0
pv_estimate10: 0
pv_estimate90: 0
- period_start: '2024-09-04T05:00:00+01:00'
pv_estimate: 0
[... 244 lines cut ...]
pv_estimate: 0
pv_estimate10: 0
pv_estimate90: 0
50dc-6601-80eb-cb70
4.25
Estimate-50dc-6601-80eb-cb70
4.25
Estimate10-50dc-6601-80eb-cb70
1.5
Estimate90-50dc-6601-80eb-cb70
6.1
D8fb-a879-bc04-d20f
6.31
Estimate-d8fb-a879-bc04-d20f
6.31
Estimate10-d8fb-a879-bc04-d20f
1.57
Estimate90-d8fb-a879-bc04-d20f
8.84
Estimate
9.5
Estimate10
2.77
Estimate90
13.45

slipx06, 2024-09-04:
Hi. Try this function

```

    filters:
      - fn: |
          ({meta}) => ({
             xs: meta.detailedForecast.map(({ period_start }) => new Date(period_start)),
             ys: meta.detailedForecast.map(({ pv_estimate }) => pv_estimate),
          })
```

Ben00793, 2024-09-05:
It's working! Thanks very much!

closed 2024-09-05 by Ben00793