slipx06/Sunsynk-Home-Assistant-Dash #27: Incorrect values of Load L1, L2, L3 Power sensors for deye_sg04lp3.yaml

SmolikOne, 2024-11-28:
Hello. I have Deye SUN-12K-SG04LP3-EU/AU integrated with your power flow card using deye_sg04lp3.yaml, but I have one issue with collected data. Namely when the solar production is delivering power above several hundred Watts per phase, values of generated power on Load L1, L2, L3 Power sensors and in Total Load Power sensor shows crazy data. So for example if PV load on phase 1, 2, 3 is lower than ca. 200W per phase, Load L1, L2, L3 Power sensors are showing correct data of power usage per particular phase. But when PV Load on phase 1, 2, 3 is higher than ca. 200W per phase, then Load sensors are showing 65.46kW load per phase which is of course incorrect. The same applies to Total Load Power sensor, that is showing combined over 194kW incorrect load. What is intersting Load data per phase on the inverter itself is showing correct data, so the question is why Solarman Integration is showing such absurd values on L1, L2, L3 Load? It there some place where I can correct it myself, or it is a error on Solarman Integration side? Thank you in advance.
[image]
[image]

slipx06, 2024-11-28:
Hi. I think you should log an issue with the Solarman integration. The card is displaying what the sensor provides so if your input data is incorrect they would need to fix it.

SmolikOne, 2024-11-28:
Understood, thank you for your suggestion.

closed 2025-02-05 by slipx06