slipx06/Sunsynk-Home-Assistant-Dash #28: Set Program Time?

chicagoandy, 2024-12-19:
A friendly feature request...

Has anyone solved how to programmatically change the time_slot times?   I don't see a way to do this, but I'm very new to Modbus

slipx06, 2024-12-19:
Hi. Yes it can be done but you need to create so modify the data before you send it to the inverter. Have a look here for some guidance

https://github.com/tomatensaus/DeyeSolarDesktop/blob/main/automations.yaml

closed 2024-12-19 by chicagoandy

slipx06, 2024-12-19:
@chicagoandy  You will also need to update your ESPHome config and make your registers writable

```
  - platform: modbus_controller            # 250 Prog1 Time
    use_write_multiple: true
    modbus_controller_id: inverter
    name: "${friendly_name} Prog1 Time"
    id: inverter_esphome_system_mode_time1
    register_type: holding
    skip_updates: ${settings_skipped_updates}
    address: 250
    icon: "mdi:clock"

  - platform: modbus_controller            # 251 Prog2 Time
    use_write_multiple: true
    modbus_controller_id: inverter
    name: "${friendly_name} Prog2 Time"
    id: inverter_esphome_system_mode_time2
    register_type: holding
    skip_updates: ${settings_skipped_updates}
    address: 251   
    icon: "mdi:clock"

  - platform: modbus_controller            # 252 Prog3 Time
    use_write_multiple: true
    modbus_controller_id: inverter
    name: "${friendly_name} Prog3 Time"
    id: inverter_esphome_system_mode_time3
    register_type: holding
    skip_updates: ${settings_skipped_updates}
    address: 252
    icon: "mdi:clock"

  - platform: modbus_controller            # 253 Prog4 Time
    use_write_multiple: true
    modbus_controller_id: inverter
    name: "${friendly_name} Prog4 Time"
    id: inverter_esphome_system_mode_time4
    register_type: holding
    skip_updates: ${settings_skipped_updates}
    address: 253
    icon: "mdi:clock"

  - platform: modbus_controller            # 254 Prog5 Time
    use_write_multiple: true
    modbus_controller_id: inverter
    name: "${friendly_name} Prog5 Time"
    id: inverter_esphome_system_mode_time5
    register_type: holding
    skip_updates: ${settings_skipped_updates}
    address: 254
    icon: "mdi:clock"

  - platform: modbus_controller            # 255 Prog6 Time
    use_write_multiple: true
    modbus_controller_id: inverter
    name: "${friendly_name} Prog6 Time"
    id: inverter_esphome_system_mode_time6
    register_type: holding
    skip_updates: ${settings_skipped_updates}
    address: 255
    icon: "mdi:clock"
```

chicagoandy, 2024-12-19:
Thanks.   Now I'm curious if there's a reason you left this out?

slipx06, 2024-12-19:
I never has a need to adjust this from HA.