slipx06/Sunsynk-Home-Assistant-Dash #39: Manual Time Sync toggle vs button

phoenix1589, 2025-04-17:
Hi,

Many thanks for your excellent work on this. I've been running an earlier version for a while and just got round to updating my ESP with the updated code including the time sync.

I'm confused about the use of an input boolean (toggle) to send a manual time sync to the inverter. The state in home assistant for the toggle is either on or off but your code only sends the time on toggle off->on. To do another manual sync it is necessary to toggle on->off, off->on.

It would feel more natural to have the manual time sync as an input_button, and perhaps the the ability to disable the automatic timesync (every hour) via an input_boolean.

Have I missed something with the current use of an input_boolean?

slipx06, 2025-05-01:
Hi. It was built quickly for testing. A more elegant solution is to define a script within ESPHome and then create a button that calls the script in HA on press. The idea was to choose one or the other not both i.e. comment out the automatic sync if doing manually etc

slipx06, 2025-10-22:
@phoenix1589 I've updated the config to use a button now