slipx06/Sunsynk-Home-Assistant-Dash #42: Wrong grid power and load power value for Deye 10kw - 1P

misu1lica, 2025-04-24:
Hi,
Many thanks for your excellent work. 
I I've been running an earlier version and  I am measuring a value that is not correct when I delivered to the network:
-  deye10kw Grid Power (grid_power_169) is of the same value cu deye10kw_grid_ct_power (grid_ct_power_172) and does not represent the value power delivered to the network;
- deye10kw_grid_ct_power (grid_ct_power_172) is slightly larger than  Grid Power (grid_power_169)  and does not represent the value power delivered to the network;

The correct value for red circle is 7.9kw-7.2kw=0.7kW (is not value grid_ct_power_172 or  grid_power_169).
The correct value for yellow circle is 7.2kw.

For my Deye 10kW is another address for modbus? How can I change value from red circle?
Thanks M

[image]

slipx06, 2025-04-24:
Hi. Have you defined a sensor for `essential_power` in your config.

misu1lica, 2025-04-25:
Hi,
where can I put essential_power?

slipx06, 2025-04-27:
Hi.  Its defined under he entities in the card.

[image]

misu1lica, 2025-04-27:
I solved the problem, grid_power_169 and grid_ct_power_172 had to be multiplied by 10.
Thanks.