slipx06/Sunsynk-Home-Assistant-Dash #43: Losing the Control items when using ESPHome-1P-Sunsynk-Deye.yaml with an ESP32-DevKitC and a MAX485

championc, 2025-04-29:
All the Sensors are loading fine, but if the ESP32 is rebooted, it takes absolutely ages for the Control Items to load up into Home Assistant.

My problem is that Automations adjust values, which have become unavailable, but suddenly appear maybe an hour or two later.  I have seen the Control items when browsing to my ESP32.

Is there a way to force everything to update to Home Assistant ?

championc, 2025-05-01:
What I noticed last night was that if I changed something in the ESPHOME Webview, then it appeared to update the controls into HA - which sounds nuts !!!

slipx06, 2025-05-01:
Does not sound normal. Are you seeing any errors in your ESPHome logs?

championc, 2025-05-01:
Further update

At around 19:00 I check and all Controls and Sensors were present.  I have an automation to start discharging the battery which starts at 20:00.  I checked my watch at 20:20 and the battery was not discharging because the Controls were gone again.

I just checked now, and the Controls are back again.

So I am at a total loss as to why they come and go.  The Sensors are ALWAYS present

championc, 2025-05-01:
I run ESPHome in a docker container.  Clicking on the logs shows this

[code]
2025-05-01 20:20:31,934 INFO 200 GET /version (127.0.0.1) 0.30ms
2025-05-01 20:21:01,975 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-05-01 20:21:32,014 INFO 200 GET /version (127.0.0.1) 0.34ms
2025-05-01 20:22:02,056 INFO 200 GET /version (127.0.0.1) 0.35ms
2025-05-01 20:22:32,097 INFO 200 GET /version (127.0.0.1) 0.31ms
2025-05-01 20:23:02,149 INFO 200 GET /version (127.0.0.1) 0.32ms
2025-05-01 20:23:32,188 INFO 200 GET /version (127.0.0.1) 0.33ms
2025-05-01 20:24:02,231 INFO 200 GET /version (127.0.0.1) 0.46ms
2025-05-01 20:24:32,292 INFO 200 GET /version (127.0.0.1) 0.33ms
2025-05-01 20:25:02,364 INFO 200 GET /version (127.0.0.1) 0.38ms
2025-05-01 20:25:32,401 INFO 200 GET /version (127.0.0.1) 0.31ms
2025-05-01 20:26:02,439 INFO 200 GET /version (127.0.0.1) 0.33ms
2025-05-01 20:26:32,478 INFO 200 GET /version (127.0.0.1) 0.32ms
2025-05-01 20:27:02,511 INFO 200 GET /version (127.0.0.1) 0.35ms
2025-05-01 20:27:32,583 INFO 200 GET /version (127.0.0.1) 0.32ms
2025-05-01 20:28:02,624 INFO 200 GET /version (127.0.0.1) 0.37ms
2025-05-01 20:28:32,665 INFO 200 GET /version (127.0.0.1) 0.35ms
2025-05-01 20:29:02,697 INFO 200 GET /version (127.0.0.1) 0.33ms
2025-05-01 20:29:32,737 INFO 200 GET /version (127.0.0.1) 0.43ms
2025-05-01 20:30:02,776 INFO 200 GET /version (127.0.0.1) 0.33ms
2025-05-01 20:30:32,849 INFO 200 GET /version (127.0.0.1) 0.35ms
2025-05-01 20:31:02,895 INFO 200 GET /version (127.0.0.1) 0.27ms
2025-05-01 20:31:32,932 INFO 200 GET /version (127.0.0.1) 0.37ms
2025-05-01 20:32:02,975 INFO 200 GET /version (127.0.0.1) 0.36ms
2025-05-01 20:32:33,012 INFO 200 GET /version (127.0.0.1) 0.34ms
2025-05-01 20:33:03,047 INFO 200 GET /version (127.0.0.1) 0.34ms
2025-05-01 20:33:33,079 INFO 200 GET /version (127.0.0.1) 0.30ms
2025-05-01 20:34:03,128 INFO 200 GET /version (127.0.0.1) 0.35ms
2025-05-01 20:34:33,170 INFO 200 GET /version (127.0.0.1) 0.33ms
2025-05-01 20:35:03,208 INFO 200 GET /version (127.0.0.1) 0.47ms
2025-05-01 20:35:33,260 INFO 200 GET /version (127.0.0.1) 0.36ms
2025-05-01 20:36:03,319 INFO 200 GET /version (127.0.0.1) 0.37ms
2025-05-01 20:36:33,359 INFO 200 GET /version (127.0.0.1) 0.34ms
2025-05-01 20:37:03,403 INFO 200 GET /version (127.0.0.1) 0.38ms
2025-05-01 20:37:33,438 INFO 200 GET /version (127.0.0.1) 0.34ms
2025-05-01 20:38:03,498 INFO 200 GET /version (127.0.0.1) 0.41ms
2025-05-01 20:38:33,535 INFO 200 GET /version (127.0.0.1) 0.33ms
2025-05-01 20:39:03,577 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-05-01 20:39:33,619 INFO 200 GET /version (127.0.0.1) 0.35ms
2025-05-01 20:40:03,656 INFO 200 GET /version (127.0.0.1) 0.47ms
2025-05-01 20:40:33,701 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-05-01 20:41:03,749 INFO 200 GET /version (127.0.0.1) 0.35ms
2025-05-01 20:41:33,822 INFO 200 GET /version (127.0.0.1) 0.44ms
2025-05-01 20:42:03,872 INFO 200 GET /version (127.0.0.1) 0.40ms
2025-05-01 20:42:33,925 INFO 200 GET /version (127.0.0.1) 0.27ms
2025-05-01 20:43:03,970 INFO 200 GET /version (127.0.0.1) 0.33ms
2025-05-01 20:43:34,010 INFO 200 GET /version (127.0.0.1) 0.53ms
2025-05-01 20:44:04,113 INFO 200 GET /version (127.0.0.1) 0.38ms
2025-05-01 20:44:34,214 INFO 200 GET /version (127.0.0.1) 1.21ms
2025-05-01 20:45:04,296 INFO 200 GET /version (127.0.0.1) 1.18ms
2025-05-01 20:45:34,373 INFO 200 GET /version (127.0.0.1) 0.31ms
2025-05-01 20:46:04,427 INFO 200 GET /version (127.0.0.1) 0.27ms
2025-05-01 20:46:34,493 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-05-01 20:47:04,555 INFO 200 GET /version (127.0.0.1) 0.33ms
2025-05-01 20:47:34,616 INFO 200 GET /version (127.0.0.1) 0.33ms
2025-05-01 20:48:04,716 INFO 200 GET /version (127.0.0.1) 0.46ms
2025-05-01 20:48:34,801 INFO 200 GET /version (127.0.0.1) 0.33ms
2025-05-01 20:49:04,873 INFO 200 GET /version (127.0.0.1) 0.40ms
2025-05-01 20:49:34,943 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-05-01 20:50:05,020 INFO 200 GET /version (127.0.0.1) 0.37ms
2025-05-01 20:50:35,080 INFO 200 GET /version (127.0.0.1) 0.37ms
2025-05-01 20:51:05,143 INFO 200 GET /version (127.0.0.1) 0.32ms
2025-05-01 20:51:35,191 INFO 200 GET /version (127.0.0.1) 0.33ms
2025-05-01 20:52:05,236 INFO 200 GET /version (127.0.0.1) 0.36ms
2025-05-01 20:52:35,304 INFO 200 GET /version (127.0.0.1) 0.32ms
2025-05-01 20:53:05,354 INFO 200 GET /version (127.0.0.1) 0.33ms
2025-05-01 20:53:35,395 INFO 200 GET /version (127.0.0.1) 0.33ms
[... 11 lines cut ...]
2025-05-01 20:59:36,026 INFO 200 GET /version (127.0.0.1) 0.34ms
2025-05-01 21:00:06,063 INFO 200 GET /version (127.0.0.1) 0.34ms
2025-05-01 21:00:36,103 INFO 200 GET /version (127.0.0.1) 0.45ms
2025-05-01 21:01:06,144 INFO 200 GET /version (127.0.0.1) 0.31ms
2025-05-01 21:01:36,179 INFO 200 GET /version (127.0.0.1) 0.44ms
2025-05-01 21:02:06,236 INFO 200 GET /version (127.0.0.1) 0.35ms
2025-05-01 21:02:36,283 INFO 200 GET /version (127.0.0.1) 0.39ms
2025-05-01 21:03:06,355 INFO 200 GET /version (127.0.0.1) 0.28ms
2025-05-01 21:03:36,424 INFO 200 GET /version (127.0.0.1) 2.07ms
2025-05-01 21:04:06,522 INFO 200 GET /version (127.0.0.1) 0.33ms
2025-05-01 21:04:36,577 INFO 200 GET /version (127.0.0.1) 0.36ms
2025-05-01 21:05:06,652 INFO 200 GET /version (127.0.0.1) 0.41ms
2025-05-01 21:05:36,694 INFO 200 GET /version (127.0.0.1) 0.32ms
2025-05-01 21:06:06,749 INFO 200 GET /version (127.0.0.1) 2.90ms
2025-05-01 21:06:36,789 INFO 200 GET /version (127.0.0.1) 0.31ms
2025-05-01 21:07:06,839 INFO 200 GET /version (127.0.0.1) 0.29ms
[... cut ...] :07:36,882 INFO 200 GET /version (127.0.0.1) 0.35ms
2025-05-01 21:08:06,945 INFO 200 GET /version (127.0.0.1) 0.31ms
2025-05-01 21:08:36,992 INFO 200 GET /version (127.0.0.1) 0.91ms
2025-05-01 21:09:07,046 INFO 200 GET /version (127.0.0.1) 0.32ms
2025-05-01 21:09:37,100 INFO 200 GET /version (127.0.0.1) 0.31ms
2025-05-01 21:10:07,161 INFO 200 GET /version (127.0.0.1) 0.47ms
[/code]

I know I manually started discharging at 20:23, and so all of the controls returned some time between then and now

slipx06, 2025-05-01:
Open the ESPHome Dashboard and check those logs 

[image]

championc, 2025-05-01:
Mine isn't online :(

I rechecked 10 mins ago, and in HA I can only see 30 Control items, and it EXCLUDES Work Mode

But when I browse to the ESP32, I can see my Deye Work Mode

So it is 100% a problem of the ESP32 sending the Control Items to HA

concierge-at, 2025-05-18:
try to use other pins like this 

`uart:
  id: mod_bus
  #tx_pin: GPIO1 #avoid
  #rx_pin: GPIO3 #avoid
  tx_pin: GPIO17  # TX-Pin = Di
  rx_pin: GPIO16  # RX-Pin = RO
  #tx_pin: GPIO22
  #rx_pin: GPIO21
  baud_rate: 9600
  stop_bits: 1
 
modbus:
  id: inverter_modbus
  # Flow Control
  flow_control_pin: GPIO4`

i use the code on esp32dev and nodemcu-32s

works fine only contact problems due to patch cables