slipx06/Sunsynk-Home-Assistant-Dash #50: Atom S3 Lite doesn't read inverter data

juanka99, 2025-09-09:
Hi,

I've been trying all afternoon and haven't been able to get it to read data from the inverter.

Do G1 and G2 correspond to GPIO1 and GPIO3?

Thanks

[image]

[image]

juanka99, 2025-09-16:
For several days now, I've been trying to flash the Atom S3 Lite to use it with the Tail485, but I'm not able to. 
Could anyone who's using them send me the .yaml file with the exact, working configuration? 
Thank you very much.

slipx06, 2025-09-16:
@edwardgarza can you help?

edwardgarza, 2025-09-18:
They are gpio pins 1 and 2. here is the relevant part of my yaml config for the atom lite

uart:
  id: mod_bus
  **tx_pin: GPIO2**
  **rx_pin: GPIO1**
  baud_rate: 9600
  stop_bits: 1
 
modbus:
  id: inverter_modbus
  flow_control_pin: GPIO16

[I can't remember if i needed this flow control pin or not but I believe I don't]

edwardgarza, 2025-09-19:
Awesome - thanks for confirming 😄